steveicarus/iverilog #1211: always_comb has glitchy behavior

ArcSpecter, 2025-02-18:
Let me just show with some code fragments where the issue is. I do have some FSM composed of two parts, combinational logic coded in always_comb and the state register transition logic coded as always_ff.

See below.
`
      // Local signals for FSM
      wire xbar_s_wready_local;
      wire xbar_m_wvalid_local;

      assign xbar_m_wvalid_local = xbar_m_wvalid[sw_req_mst];
      assign xbar_s_wready_local = xbar_s_wready[slave_index];

      // XBAR Serve FSM
      xbar_state_t s_current, s_next;  // State registers

      always_comb begin : xbar_fsm_comb
        // Default output
        s_next = s_current;
        sw_xfer_cnt_en=0;

        // Regs
        sw_xfer_cnt = sw_xfer_cnt;

        case (s_current)
          S_IDLE: begin
            s_next = S_QUEUE_WAIT;

            // Init regs
            sw_xfer_cnt = 0;
          end

          S_WRITE: begin
            // Redirection to other signals
            xbar_s_wvalid[slave_index] = xbar_m_wvalid_local;
            xbar_m_wready[sw_req_mst] = xbar_s_wready_local;

            // if (xbar_s_wready[slave_index] && xbar_m_wvalid[sw_req_mst]) begin // THIS IS SAME AS the _local version of signal but gives different glitch
            if (xbar_s_wready_local && xbar_m_wvalid_local) begin // THIS CONDITION IS NOT TRUE
              sw_xfer_cnt_en=1; // AS A RESULT THIS NEVER SETS TO 1
              sw_xfer_cnt = sw_xfer_cnt + 1; // BUT THIS ONE SETS FROM 0 TO 2 (!)
            end

            // Irrelevant code omitted
          end

          default: begin
            s_next = S_IDLE;
          end

        endcase
      end

      always_ff @(posedge clk) begin : xbar_fsm_reg
        if (!rst_n) begin
          s_current <= S_IDLE;
        end else begin
          s_current <= s_next;
        end
      end`

Inside a S_WRITE state there is very simple comparison of two signals, they both need to be 1 in order for the sw_xfer_cnt_en to turn to 1 as well as in order for sw_xfer_cnt to increment (since originally in S_IDLE state its reset to 0).

The whole thing is located inside a generate statement and slave_index==1.

This is simple, and in this current simulation, the compared signals are not 1 at the same time, i.e. they do switch, thus, the condition should NOT be met and whatever inside that if statement should NOT execute in any way.

Well, above is observed for the sw_xfer_cnt_en signal (it stays 0 as after being initialized), HOWEVER; the issue is that I do see that the sw_xfer_cnt signal IS being incremented, and not even by one but by 2. See the attached waveforms, you can see that it turns from 0 to 2, whereas it should not have done that (just as sw_xfer_cnt_en which stayed 0) because looking at the xbar_s_wready and xbar_s_wvalid they are not 1 at the same time, theres switching between them which results in not meeting this condition. 

so question: Why does sw_xfer_cnt is changing at all?
Also, you see it increments later on too, this time increments properly by 1, from 2 to 3, but again it shouldnt because those condition signals as per what you see in waveform are not 1 at same time.

One more thing you noticed is, I do compare the assigned "_local" versions of those signals, and assignment is done outside of this FSM. The reason for that is, if I use those signals in "if" condition (as shown in commented out code) then I get warning:
`sorry: constant selects in always_* processes are not currently supported (all bits will be included).`

I am assuming it is limitation of Iverilog, thus I do create the "_local" versions of those signals before using them in comparison. 

However, interesting thing is, if I still ignore warning and use bare signals, the sw_xfer_cnt jumps to 4 from 2 (which it shouldnt either)

so what is going on here?

BTW: One way to "mitigate" this is, if i take out sw_xfer_cnt outside of this FSM and just create regular always_ff blocked counter controlled by sw_xfer_cnt_en from FSM, then it works, i.e. whenever condition NOT met that sw_xfer_cnt is not incremented because sw_xfer_cnt_en never toggles (as it should) and when condition met, it just incremented by 1.

So, any comments?

[image]
[image]

martinwhitaker, 2025-02-18:
Please provide a complete test case that we can run to reproduce this. Reduce it to the minimum code that demonstrates the problem.

ArcSpecter, 2025-02-19:
> First always_comb blocks are expected to have glitches since they execute each time a variable changes not just once. The always_ff is used to make sure the current values are only updated at the clock edge.

Yes, once I changed my FSM type to single part FSM with always_ff (and reworked logic a bit), then it was working as expected.

The key part is "always_comb blocks are expected to have glitches since they execute each time a variable changes".

closed 2025-02-19 by ArcSpecter