steveicarus/iverilog #1213: Assertion failure in `ivl_signal_nex` with `uwire` array port

L3odr0id, 2025-02-28:
Compiling the following SystemVerilog module throws an assertion:

```
module a(output uwire o1 [0:1]);
endmodule
```

Command used:
```
iverilog -g2012 test.sv
```

Error output:
```
ivl: t-dll-api.cc:2501: ivl_nexus_s* ivl_signal_nex(ivl_signal_t, unsigned int): Assertion `net->type_ == IVL_SIT_REG' failed.
Aborted
```

This issue occurs in `Icarus Verilog version 13.0 (devel) (s20221226-572-gb74511974)`

martinwhitaker, 2025-03-08:
This only seems to occur if
a) the module is a top-level module
and
b) the output port is undriven

caryr, 2025-03-19:
Is this example the result of a fuzzing framework?

L3odr0id, 2025-03-19:
Actually it's not a fuzzing framework, but a property-based testing library. This example was found by a [verilog model](https://github.com/DepTyCheck/verilog-model) using [deptycheck](https://github.com/buzden/deptycheck). The generator returned a long module, and I shrinked manually to get a minimal example that reproduced the error