steveicarus/iverilog #1215: Ternary expressions affect assignment errors

FSY369, 2025-03-07:
Hi, I'm running the following code to assign a value to wire11 and I'm getting an error.
`assign wire11 = ((7'h40) ?
                      $signed($unsigned({((7'h43) >>> wire0)})) : ((8'hb1) <<< wire1));`
Normally, (conditional expression) ? (branch 1) : (branch 2).
Since 7'h40 is always greater than 1 and non-zero values are considered true in the conditional expression, branch 1 is run, so the assignment to wire11 can be abbreviated again
`assign wire11 = $signed($unsigned((7'h43 >>> wire0)));`
 but emulating each of them finds an inconsistency in the 130th column of the result of the y-value.

[image]

caryr, 2025-03-07:
I thought I posted a comment with this, but it's missing. We need a simple working example that demonstrates the issue. A single line is not enough since there are other variables in the expression.

FSY369, 2025-03-10:
Ok, I'll provide rtl.v and the stimulus file testbench, hope that helps you!
[rtl.v.log](https://github.com/user-attachments/files/19153702/rtl.v.log)

[rtl.v.log](https://github.com/user-attachments/files/19153705/rtl.v.log)

caryr, 2025-03-15:
This is likely the same issue as #1216. Please make sure you understand how the width and signed information is handled by the various expression operators.

FSY369, 2025-03-19:
Thank you very much for answering my query, sorry for the lack of understanding before, but I found another problem, when I simulated this use case on vivado, I found that the y-values were inconsistent.

caryr, 2025-03-19:
Is it an Icarus or Vivaldo issue? There are a number of cases where Icarus reports the value the standard dictates while the commercial simulators report an incorrect value. The only way you can figure this out is to break this down to the individual pieces and check these against the standard. Even the examples you provided earlier have a significant amount of extra code that obscures the fundamental issue. To resolve this we need a concise example that demonstrates the issue. This is often just the result of refactoring as you check against the standard.

closed 2025-06-19 by caryr