steveicarus/iverilog #1216: Ternary expression causes bit width truncation

FSY369, 2025-03-07:
Hi, I'm running the following code to assign a value to wire23 and I'm getting an error.
` assign wire23 = ((8'haa) ?
                      $signed((($signed((8'ha8)) << (~&reg18)) ~^ reg18[(1'h1):(1'h0)])) : reg18[(2'h3):(2'h2)]);  `
Normally, (conditional expression) ? (branch 1) : (branch 2).
Since 8'haa is always greater than 1 and non-zero values are considered true in the conditional expression, branch 1 is run, so the assignment to wire23 can again be abbreviated to the following code:
` assign wire23 =
                      $signed((($signed((8'ha8)) << (~&reg18)) ~^ reg18[(1'h1):(1'h0)])); `
by definition they both simulate the same result, but the ternary expression bit-width truncation leads to an error.

caryr, 2025-03-07:
I thought I posted a comment with this, but it's missing. We need a simple working example that demonstrates the issue. A single line is not enough since there are other variables in the expression.

FSY369, 2025-03-10:
Ok, I'll provide rtl.v and the stimulus file testbench, hope it helps you!
[rtl.v.log](https://github.com/user-attachments/files/19153787/rtl.v.log)

[identity_testbench.v.log](https://github.com/user-attachments/files/19153789/identity_testbench.v.log)

caryr, 2025-03-15:
Icarus is generating the correct result. Your misunderstanding is likely that the ternary operator can modify both the expression width and the signed/unsigned nature of the expression. What you have in the code is effectively `1'b1 ? $signed(<value>) : <unsigned constant>` since the second expression is unsigned the entire expression is unsigned so it is actually equivalent to `assign wire23 = $unsigned($signed((($signed((8'ha8)) << (~&reg18)) ~^ reg18[(1'h1):(1'h0)])));` not what you stated previously. The expression is expected to zero extended not sign extended. If you want signed extension then the second term would also have to be cast to a signed result or the entire ternary operator needs to be cast to signed.

closed 2025-06-19 by caryr