steveicarus/iverilog #1218: Simulation of a continuous assignment of an unpacked array throws error

L3odr0id, 2025-03-14:
Simulation of the following SystemVerilog module throws an error:

```
module a (output logic b [2:1]);
  assign b = '{'b1,'bz};
endmodule: a
```

Commands:
```
iverilog -g2012 test.sv -o a.out
vvp a.out
```

Error output:
```
Unable to resolve label v0x55abf7187e10_0
compile_cleanup: 1 unresolved items
a.out: Program not runnable, 1 errors.
```

Expected behavior:
The module both compiles and simulates successfully or a human-readable error is thrown

This issue occurs in `Icarus Verilog version 13.0 (devel) (s20221226-572-gb74511974)`

caryr, 2025-03-19:
Is this example the result of a fuzzing framework?

L3odr0id, 2025-03-19:
Found by a [verilog model](https://github.com/DepTyCheck/verilog-model) using [deptycheck](https://github.com/buzden/deptycheck)

caryr, 2025-03-19:
Ok, it seemed like fuzzing since all the reports are related to unconnected/unused top level ports which is an obscure corner case. Are these intended to be general functional models, but you do not have a test bench in the simulation?

L3odr0id, 2025-03-20:
Well they aren't meant to be a functional model. These models were generated to check if different Verilog tools follow the IEEE specification