steveicarus/iverilog #1219: Invalid array dimension declaration causes assertion failure and crash

FSY369, 2025-03-18:
Steps to Reproduction：
1. crash.sv：
  `module crash;
     bit invalid_arr [0:-1]; 
     assign invalid_arr = '{}; 
   endmodule` 
2. compile command：
`iverilog -g2012 crash.sv`
3. actual output：
`crash.sv:2: assert: elaborate.cc:214: failed assertion rval_pident
Aborted (core dumped)`
4. expected behaviour：
`Output explicit error message (e.g. ‘Invalid array dimension range [0:-1]’)`

caryr, 2025-06-20:
The latest development reports the array to array assignment pattern size mismatch.

caryr, 2025-06-30:
There was a segfault since the code was not checking for the elaboration failure of the empty array concatenation before trying to build the r-value expression code. A fix for this was pushed to development.  I'll close this now since no new information has been provided.

closed 2025-06-30 by caryr