steveicarus/iverilog #1225: Iverilog freezes in invalid infinite loop

joaovam, 2025-03-26:
Icarus freezes when analyzing an infinite loop instead of reporting an error or terminating normally.

### Design:
```systemverilog
module module_0;
   genvar id_6;

   for (genvar id_8 = -1; -1; id_6 = -1);
endmodule
```

Iverilog version: `Icarus Verilog Parser/Elaborator version 13.0 (devel) (s20221226-554-g25a84d5cf)`
Design generated by Fuzzing.

caryr, 2025-05-28:
There is also an error with the increment expression since the standard requires that the initial expression assignment and the increment expression must be to the same genvar.

caryr, 2025-06-22:
I have pushed a fix for this to the development branch. It reports a large number of errors. Though there is an issue in msys that needs a minor fix. That I will resolve next.

closed 2025-06-22 by caryr