steveicarus/iverilog #1230: Parameters cannot be used in RHS of input port (default value)

jmi2k, 2025-03-31:
Those snippets of code fail under the latest version of Icarus Verilog (both stable 12.0 and development `f5decd4`).

```sv
module Foo #(parameter F) (input logic i = F);
endmodule

module Test;
	Foo #(.F(0)) f();
endmodule
```

```sv
module Foo #(parameter F) (input logic i = F'(1));
endmodule

module Test;
	Foo #(.F(0)) f();
endmodule
```

Both result in:

```
$ iverilog -v -g 2012 test/sandbox.sv
[...]
Using language generation: IEEE1800-2012,no-specify,no-interconnect,xtypes,icarus-misc
PARSING INPUT
LOCATING TOP-LEVEL MODULES
   Test
 ... done, 0 seconds.
ELABORATING DESIGN
test/sandbox.sv:1: error: Unable to bind parameter `F' in `Test'
test/sandbox.sv:1: error: Failed to elaborate port default value.
2 error(s) during elaboration.
```

However, something like this works (which is not equivalent, it's just setting the initial value of a register, but syntactically it looks similar enough to make me expect consistent behavior):

```sv
module Foo #(parameter F) (output logic o = F);
endmodule

module Test;
	Foo #(.F(0)) f();
endmodule
```

caryr, 2025-06-20:
I verified this is still failing using the latest development.

caryr, 2025-07-08:
FYI this is because the elaboration of the default expression is being done in the wrong scope. Normally input expressions are elaborated in the calling scope, but for default expressions they need to be elaborated in the module instance scope. This is not too hard to implement, but it needs to be done on a per instance basis which is complicated by module array instantiation where the individual instances can have different expression results.

I have a plan on how to make all this work that I have verified works for the single instance case shown above, but need some free time to expand this to the general multiple instance case.

jmi2k, 2025-07-08:
Sure, thanks for having a look at it!