steveicarus/iverilog #1233: Include files passed to `$readmem[bh]` to the list of files that contribute to the compilation of the design (`-M`)

jmi2k, 2025-04-11:
The list output by `-M` is rather useful to make a `.d` file `make` can include. Yosys provides such a facility, but as opposed to Icarus Verilog it includes paths passed to `$readmemh` and `$readmemb`. This is great, as those files also need to be considered as dependencies and rebuilt accordingly.

martinwhitaker, 2025-04-13:
The problem with this request is that these files don't contribute to the compilation of the design. The `$readmem` tasks, like most other system tasks and functions, are handled by the `vvp` runtime.

I'll leave this open as an enhancement request, but think it unlikely it will get implemented.

I would think you could get the information you need just as easily using `grep` or `awk`.

jmi2k, 2025-04-13:
Does it make sense to add such behavior to `vvp` then?