steveicarus/iverilog #1235: Inconsistent reg update on negedge due to input assignment

sdjasj, 2025-04-21:
The following code triggers the issue:

```verilog
`timescale 1ns/1ps
module tb_a;

    reg  [26:0] in0;
    reg         clock_0;
    reg         clock_1;
    wire        out0;

    a dut (
        .in0     (in0),
        .clock_0 (clock_0),
        .clock_1 (clock_1),
        .out0    (out0)
    );

    initial begin

        in0     = 27'd0;
        clock_0 = 1'b0;
        clock_1 = 1'b0;

        #2000;
        in0     = 27'd0;
        clock_0 = 1'b1;    
        clock_1 = 1'b0;

        #2000;
        in0     = 27'd1;
        clock_0 = 1'b0;   // 1 -> 0, negedge
        clock_1 = 1'b0;

        #2000;
        $finish;
    end

    initial begin
        $monitor("t=%0t ns  in0=%0d clk0=%b clk1=%b | out0=%b",
                 $time, in0, clock_0, clock_1, out0);
    end

endmodule

module a (in0, clock_0, clock_1, out0);

input wire [26:0] in0;
wire [3:0] wire_0;
reg reg_0;
output wire out0;
input wire clock_0;
input wire clock_1;

    initial begin
        #1;
        reg_0 = 0;
    end

    assign wire_0 = in0[0:0];
    assign out0 = reg_0;
    
    always @(negedge clock_0 or posedge clock_1) begin
      if (clock_1) begin
        reg_0 <= 1;
      end else begin
        reg_0 <= wire_0;
      end
    end
endmodule
```

You can reproduce the issue with the following commands:
```
iverilog tb.v -o a.out
./a.out
```

**Output from Icarus Verilog (iverilog):**
```
t=0 ns  in0=0 clk0=0 clk1=0 | out0=x
t=1000 ns  in0=0 clk0=0 clk1=0 | out0=0
t=2000000 ns  in0=0 clk0=1 clk1=0 | out0=0
t=4000000 ns  in0=1 clk0=0 clk1=0 | out0=0
tb.v:34: $finish called at 6000000 (1ps)
```

**Output from Verilator:**
```
t=0 ns  in0=0 clk0=0 clk1=0 | out0=0
t=2000000 ns  in0=0 clk0=1 clk1=0 | out0=0
t=4000000 ns  in0=1 clk0=0 clk1=0 | out0=0
t=4000000 ns  in0=1 clk0=0 clk1=0 | out0=1
- tb.v:35: Verilog $finish
- S i m u l a t i o n   R e p o r t: Verilator 5.035 devel
- Verilator: $finish at 6us; walltime 0.044 s; speed 75.933 us/s
- Verilator: cpu 0.079 s on 1 threads; alloced 14681163 MB
```

This seems to be caused by the value of `in0` not being updated in time during the falling edge of `clock_0`, resulting in `reg_0` not capturing the correct value.

martinwhitaker, 2025-04-21:
This is an obvious race condition in your code. Ask yourself this question: will the assignment `wire_0 = in0[0:0]` be executed before or after the assignment `reg_0 <= wire_0`? Then read the "Scheduling semantics" section in the LRM (IEEE 1364 or IEEE 1800) to discover why either result is valid.

sdjasj, 2025-04-21:
Thank you for pointing that out. I did overlook the impact of scheduling order on simulation results. I’ll reread the relevant section in the LRM about scheduling semantics to avoid similar race conditions. Thanks again for your detailed explanation!

closed 2025-04-21 by sdjasj