steveicarus/iverilog #1237: Case statement mismatch with signed 1-bit reg compared to unsigned constant

sdjasj, 2025-04-28:
Below is the code that triggers the problem (`test.v`):

```verilog
module test;

reg signed in4;
initial begin
  in4 = 1;

  #1;

  $display("in4: %0d", in4);

  case (0 ? 1'h0 : in4)
    5'b0101: begin
        $display("1");
    end

    8'b000001: begin
      $display("2");
    end

    default: begin
      $display("3");
    end
  endcase
end

endmodule
```

Run the commands below to reproduce:

```bash
iverilog test.v
./a.out
```

The program prints

```
in4: -1
3
```

According to my understanding, the ternary operator should choose `in4`, because `0` is `false`.  
`in4` is a **signed** value and its binary representation is `1'b1`.  
However, when `in4` is compared with the unsigned value `8'b000001`, it is treated as unsigned (as explained here: [https://stackoverflow.com/questions/21340093/why-is-this-verilog-relational-statement-returning-true](https://stackoverflow.com/questions/21340093/why-is-this-verilog-relational-statement-returning-true)).  
Thus, `1'b1` is zero-extended to `8'b0000_0001`, making it equal to the second case branch.  
Therefore, it should print `2` instead of `3`.

caryr, 2025-06-23:
I started looking at this. If the code passed to `vvp` is hacked to use unsigned padding instead of signed padding then this works as expected so it's a problem in the `iverilog` stage. The vlog95 target result is also missing the cast to unsigned after the ternary operator is optimized away so it's not in the code generation. This implies it's in the elaboration or optimization stages.

I'll look at this more when I have some free time.

caryr, 2025-06-25:
The case statement is pruned to make the condition and the case labels as small as possible. The prune code was improperly using the sign of the base expression instead of the original padded expression when resizing the condition.

A fix for this has been push to the development branch.