steveicarus/iverilog #1238: mux assert failure

dd1dd1, 2025-04-30:
hi, i am using iverilog as a quick syntax checker for a big xilinx vivado project, this works very well, I greatly appreciate your work! 5 seconds vs 5 minutes to find a missing comma makes a big difference.

I would like to report one failure case where vivado is happy and code works, but ivl aborts with an error. I am not 100% sure if my code is valid verlog. IANAL.

common/hdl/ds20k/pkt16_decode.sv:101: error: Cannot evaluate wc in this context.
common/hdl/ds20k/pkt16_decode.sv:101: assert: net_func_eval.cc:1048: failed assertion sub_exp
Aborted (core dumped)

error line reads "bitmux[wc] <= sink16_data" and expands into a data multiplexor.

this is the complete file:

daq00:ds-dm-gcdm$ more common/hdl/ds20k/pkt16_decode.sv
//
// receive a 16-bit packet. K.O. Nov 2024
//

`default_nettype none
module pkt16_decode
  #(parameter N = 64)
  (
   input  wire        clk,
   input  wire        reset,

   input  wire [7:0]  packet_type,

   input  wire [15:0] sink16_data,
   input  wire        sink16_eop,
   input  wire        sink16_valid,

   output reg [N-1:0] data_out,
   output reg         done_out,
   output reg         error_out
   );

   localparam N1  = N-1;
   localparam NW  = N/16;
   localparam NW1 = NW-1;

   genvar             i;

   reg [NW1:0][15:0]    bitmux;
   
   reg [N1:0]         data;
   
   for (i=0; i<NW; i++) begin
      assign data[i*16+15:i*16+0] = bitmux[i];
   end

   reg in_packet;
   reg our_packet;
   reg [7:0] wc;

   always_ff @(posedge clk) begin
      if (reset) begin
         in_packet  <= 0;
         our_packet <= 0;
         wc         <= 0;
         bitmux     <= 0;
         data_out   <= 0;
         done_out   <= 0;
         error_out  <= 0;
      end else if (sink16_valid) begin
         if (!in_packet) begin // first word of a packet
            if (sink16_eop) begin // unexpected eop
               in_packet    <= 0;
               our_packet   <= 0;
               data_out     <= 16'hFFFC;
               done_out     <= 0;
               error_out    <= 1;
            end else begin
               in_packet    <= 1;

               if (sink16_data[7:0] == packet_type) begin // this packet is for us
                  our_packet   <= 1;
                  bitmux[0]    <= sink16_data;
                  wc           <= 1;
                  //data_out     <= 0;
                  done_out     <= 0;
                  error_out    <= 0;
               end else begin
                  our_packet   <= 0;
                  //data_out     <= 0;
                  done_out     <= 0;
                  error_out    <= 0;
               end
            end
         end else begin // in packet
            if (our_packet) begin
               if (wc == NW) begin // expect end of packet
                  if (sink16_eop) begin // end of packet, all good
                     wc         <= 0;
                     data_out   <= data;
                     done_out   <= 1;
                     error_out  <= 0;
                     our_packet <= 0;
                     in_packet  <= 0;
                  end else begin // packet is too long
                     wc         <= 0;
                     data_out   <= 16'hFFFE;
                     done_out   <= 0;
                     error_out  <= 1;
                     our_packet <= 0; // ignore remaining data
                  end
               end else begin
                  if (sink16_eop) begin // unexpected early end of packet, packet is too short
                     wc         <= 0;
                     data_out   <= 16'hFFFD;
                     done_out   <= 0;
                     error_out  <= 1;
                     in_packet  <= 0;
                     our_packet <= 0;
                  end else begin
                     bitmux[wc] <= sink16_data;
                     wc         <= wc + 1;
                     //data_out   <= 0;
                     done_out   <= 0;
                     error_out  <= 0;
                  end
               end
            end else begin // ignore other packets
               if (sink16_eop) begin // end of packet
                  in_packet  <= 0;
                  our_packet <= 0;
               end
               //data_out   <= 0;
               done_out   <= 0;
               error_out  <= 0;
            end
         end
      end else begin // if (sink16_valid)
         //data_out   <= 0;
         done_out   <= 0;
         error_out  <= 0;
      end
   end
   
endmodule

dd1dd1, 2025-04-30:
I am using latest pull of iverilog and I can make a reduced example source code that shows this problem if it is helpful for you.

caryr, 2025-07-01:
This code compiles using the latest stable and development branches without any errors. Can you please verify if this is still an issue.