steveicarus/iverilog #1239: packed array error

dd1dd1, 2025-04-30:
hi, i am using iverilog as a quick syntax checker for a big xilinx vivado project, this works very well, I greatly appreciate your work! 5 seconds vs 5 minutes to find a missing comma makes a big difference.

I would like to report another failure case where vivado is happy and code works, but ivl reports an error. I am not 100% sure if my code is valid verlog. IANAL.

Here is the example errors:
```
common/hdl/ds20k/gdm_busy_veto.sv:50: warning: @* is sensitive to all 12 words in array 'qsfp_rx_bsy_counter'.
common/hdl/ds20k/gdm_busy_veto.sv:51: warning: @* is sensitive to all 12 words in array 'qsfp_rx_bsy_counter'.
common/hdl/ds20k/gdm_busy_veto.sv:56: warning: @* is sensitive to all 12 words in array 'qsfp_rx_bsy_counter'.
common/hdl/ds20k/gdm_busy_veto.sv:44: warning: 'qsfp_rx_bsy_counter' is driven by more than one process.
common/hdl/ds20k/gdm_busy_veto.sv:44: warning: 'qsfp_rx_bsy_counter_latched_out' is driven by more than one process.
common/hdl/ds20k/gdm_busy_veto.sv:44: sorry: Cannot synthesize signals that are driven by more than one process.
```

Here is the source code and full error messages:
```
daq00:ds-dm-gcdm$ cat common/hdl/ds20k/gdm_busy_veto.sv
`default_nettype none
  
module gdm_busy_veto
  (
   input wire        clk,
   input wire        reset,
   input wire [63:0] ts64,
   input wire        cmd_latch,
   input wire [11:0] qsfp_rx_bsy,
   input wire        qsfp_bsy_force,
   input wire [11:0] qsfp_mask,
   input wire [15:0] vx_bsy_extend,
   input wire [15:0] gdm_veto_extend,
   output reg        gdm_veto_out,
   output reg        gdm_veto_pulse_out,
   output reg [15:0] gdm_veto_pulse_counter_latched_out,
   output reg [15:0] gdm_veto_up_counter_latched_out,
   output reg        gdm_busy_encode_error_latch_out,
   output reg [15:0] pkt16_data_out,
   output reg        pkt16_eop_out,
   output reg        pkt16_valid_out,
   input wire        pkt16_ready,
   output reg        gdm_bsy_out,
   output reg [15:0] gdm_bsy_up_counter_latched_out,
   output reg [15:0] gdm_bsy_pulse_counter_latched_out,
   output reg [15:0] gdm_bsy_refresh_counter_latched_out,
   output reg [7:0]  qsfp_rx_bsy_counter_latched_out[11:0],
   output reg [31:0] gdm_bsy_status_latched_out
   );

   reg               gdm_veto;

   //////////////////////////////////////////////////////////////////
   // GDM busy logic
   //////////////////////////////////////////////////////////////////

   genvar     iqsfp;

   // counters 0 to 255, reset at run start
   
   reg [7:0]  qsfp_rx_bsy_counter[11:0];

   for (iqsfp = 0; iqsfp < 12; iqsfp++) begin
      always_ff @(posedge clk) begin
         if (reset) begin
            qsfp_rx_bsy_counter[iqsfp] <= 0;
            qsfp_rx_bsy_counter_latched_out[iqsfp] <= 0;
         end else begin
            if (qsfp_rx_bsy[iqsfp]) begin
               if (qsfp_rx_bsy_counter[iqsfp] != 8'hFF) begin
                  qsfp_rx_bsy_counter[iqsfp] <= qsfp_rx_bsy_counter[iqsfp] + 1;
               end
            end
[... 398 lines cut ...]
common/hdl/ds20k/cdm_busy_veto.sv:52: warning: @* is sensitive to all 12 words in array 'vx_bsy_counter'.
common/hdl/ds20k/cdm_busy_veto.sv:57: warning: @* is sensitive to all 12 words in array 'vx_bsy_counter'.
common/hdl/ds20k/cdm_busy_veto.sv:45: warning: 'vx_bsy_counter' is driven by more than one process.
common/hdl/ds20k/cdm_busy_veto.sv:45: warning: 'vx_bsy_counter_latched_out' is driven by more than one process.
common/hdl/ds20k/cdm_busy_veto.sv:45: sorry: Cannot synthesize signals that are driven by more than one process.
common/hdl/ds20k/cdm_busy_veto.sv:51: warning: @* is sensitive to all 12 words in array 'vx_bsy_counter'.
common/hdl/ds20k/cdm_busy_veto.sv:52: warning: @* is sensitive to all 12 words in array 'vx_bsy_counter'.
common/hdl/ds20k/cdm_busy_veto.sv:57: warning: @* is sensitive to all 12 words in array 'vx_bsy_counter'.
common/hdl/ds20k/cdm_busy_veto.sv:45: warning: 'vx_bsy_counter' is driven by more than one process.
common/hdl/ds20k/cdm_busy_veto.sv:45: warning: 'vx_bsy_counter_latched_out' is driven by more than one process.
common/hdl/ds20k/cdm_busy_veto.sv:45: sorry: Cannot synthesize signals that are driven by more than one process.
common/hdl/ds20k/cdm_busy_veto.sv:51: warning: @* is sensitive to all 12 words in array 'vx_bsy_counter'.
common/hdl/ds20k/cdm_busy_veto.sv:52: warning: @* is sensitive to all 12 words in array 'vx_bsy_counter'.
common/hdl/ds20k/cdm_busy_veto.sv:57: warning: @* is sensitive to all 12 words in array 'vx_bsy_counter'.
common/hdl/ds20k/cdm_busy_veto.sv:45: warning: 'vx_bsy_counter' is driven by more than one process.
common/hdl/ds20k/cdm_busy_veto.sv:45: warning: 'vx_bsy_counter_latched_out' is driven by more than one process.
common/hdl/ds20k/cdm_busy_veto.sv:45: sorry: Cannot synthesize signals that are driven by more than one process.
common/hdl/ds20k/cdm_busy_veto.sv:51: warning: @* is sensitive to all 12 words in array 'vx_bsy_counter'.
common/hdl/ds20k/cdm_busy_veto.sv:52: warning: @* is sensitive to all 12 words in array 'vx_bsy_counter'.
common/hdl/ds20k/cdm_busy_veto.sv:57: warning: @* is sensitive to all 12 words in array 'vx_bsy_counter'.
common/hdl/ds20k/pkt16_decode.sv:101: error: Cannot evaluate wc in this context.
common/hdl/ds20k/pkt16_decode.sv:101: assert: net_func_eval.cc:1048: failed assertion sub_exp
Aborted (core dumped)
make: *** [Makefile:109: iverilog] Error 134
```

dd1dd1, 2025-04-30:
I am using latest pull of iverilog and I can make a reduced example source code that shows this problem if it is helpful for you.

martinwhitaker, 2025-05-01:
Synthesis support in `iverilog` is very limited, and IMO not very useful. If all you want to do is check for syntax errors, omit the `-S` option - it isn't gaining you anything.

For fast syntax checking, `-tnull` should be better than `-tvlog95`

martinwhitaker, 2025-05-11:
As far as I can see, the only error message is the one you already reported in #1238. The "sorry" messages are just saying that the Icarus synthesis option does not support that particular coding style. It can only perform flip-flop inference if all bits of a vector are driven from the same `always` process.

As I said in my previous response, the solution is to compile without the `-S` option - you don't need it.

Closing as the compiler is behaving as intended. There are no plans to extend synthesis support.

closed 2025-05-11 by martinwhitaker