steveicarus/iverilog #1241: Partial initialization of reg unexpectedly overridden with x

sdjasj, 2025-05-02:
Below is the problematic code `test.v`:

```verilog
`timescale 1ns/1ps
module qfhclxakch(in1, clock_0, clock_2, out99);
  input wire [1:0] in1;
  input wire clock_0;
  input wire clock_2;
  output wire [27:14] out99;
  wire [8:0] _133_;
  reg [21:0] reg_1;
  
  initial begin
    reg_1 = 22'h000000;
  end

  assign _133_ = in1[1] ? { 8'h00, in1[0] } : reg_1[13:4];
  assign out99 = reg_1[13:0];

  always @(negedge clock_0, posedge clock_2)
    if (clock_2) reg_1 <= 22'h0003cb;
    else reg_1 <= { reg_1[21:13], _133_, reg_1[3:0] };

endmodule
```

And the corresponding testbench file `tb.v`:

```verilog
`timescale 1ns/1ps
module tb_qfhclxakch;

  reg  [1:0] in1;
  reg        clock_0;
  reg        clock_2;
  wire [27:14] out99;

  qfhclxakch dut (
      .in1     (in1),
      .clock_0 (clock_0),
      .clock_2 (clock_2),
      .out99   (out99)
  );

  initial begin
    {in1, clock_0, clock_2} = 0;
    #2000;
    $display("reg1: %b", dut.reg_1);
    
    $finish;
  end
endmodule
```

The following commands reproduce the issue:

```
iverilog test.v tb.v 
./a.out
```

iverilog produces the following output:

```
reg1: 000000000xxxxxxxxx0000
tb.v:21: $finish called at 2000000 (1ps)
```

It can be seen that the initialization of `reg_1` in `test.v` is only partially effective, as the bits in the \[12:4] range are initialized to `x`. I suspect the issue is related to the line `reg_1 <= { reg_1[21:13], _133_, reg_1[3:0] };`, since `_133_` corresponds to that range. I'm not sure whether this behavior is expected.

caryr, 2025-05-02:
This is permissible behavior.

To understand what is going on you need to look at the output and work back to how the result is being generated. First, since some bit are set to zero we know the the initial block to initialize reg_1 was executed so it was likely all 0 at some point. You did correctly notice that the issue seems to be related to the assignment in the always block which is triggered at time-0 when clock_0 transitions from an X to a 0 (which is defined to be a negative edge). Now the question would be why does _133_ have an unknown value? The issue is there is a race between the negative edge of the clock triggering the always block and the continuous assignment generating the expected result for _133_. 

It's usually wrong to assume the simulator is not function correctly until you have verified there is not something wrong with your code or test bench and for cases like this it is almost always some race condition that is not understood. Make sure you understand that there is no guaranteed order for initial, always, or continuous assignments. Though most simulators do try to run always blocks before initial blocks so they can be properly sensitized for any changes that happen in initial block.

sdjasj, 2025-05-02:
I understand why now—thank you for your detailed and patient reply :)

closed 2025-05-02 by sdjasj