steveicarus/iverilog #1243: Incorrect Result for `0 ** x` Expression: Should Be 0 but Produces Unknown (`x`)

sdjasj, 2025-05-11:
Here is the problematic code `test.v`:

```verilog
`timescale 1ns/1ps
module top (out12);
    reg signed [17:8] wire_1 = ~(6'o76);
    output wire  [19:10] out12;

    assign out12 = wire_1[10:10] ** wire_1;

    initial begin
      #1000;
      $display("wire: %b", wire_1);
      $display("out12: %b", out12);
      $finish;
    end

endmodule
```

The issue can be reproduced with the following commands:

```
iverilog test.v
./a.out
```

The output from Icarus Verilog is:

```
wire: 1111000001
out12: xxxxxxxxxx
```

However, since `wire_1` is `10'b1111000001`, we have `wire_1[10:10] = 1'b0`, so the expression `wire_1[10:10] ** wire_1` should evaluate to `0 ** wire_1`, which is `0`.

martinwhitaker, 2025-05-11:
`wire_1` is a signed value, so `10'b1111000001` is a negative value. From 1364-2005 section 5.1.5:

>If neither operand of the power operator is real, then the result type shall be determined as outlined in 5.4.1
and 5.5.1. The result value is 'bx if the first operand is zero and the second operand is negative. The result
value is 1 if the second operand is zero.

So Icarus Verilog is producing the correct result.

sdjasj, 2025-05-11:
Thank you for your response, and I apologize for overlooking the signedness of `wire_1`. I'm sorry for wasting your time.

closed 2025-05-11 by sdjasj