steveicarus/iverilog #1245: Error trying to compile iverilog

jordankrim, 2025-05-23:
The is the error I get on RHEL8 with gcc 14.2.0:

>make check
g++ -DHAVE_CONFIG_H -I. -Ilibmisc  -Wall -Wextra -Wshadow   -g -O2 -std=c++11 -MD -c main.cc -o main.o
main.cc: In function 'int main(int, char**)':
main.cc:985:40: error: 'VERSION_TAG' was not declared in this scope; did you mean 'VERSION_STR'?
  985 |                  << VERSION << " (" << VERSION_TAG << ")" << endl << endl;
      |                                        ^~~~~~~~~~~
      |                                        VERSION_STR
main.cc:998:40: error: 'VERSION_TAG' was not declared in this scope; did you mean 'VERSION_STR'?
  998 |                  << VERSION << " (" << VERSION_TAG << ")"  << endl <<
      |                                        ^~~~~~~~~~~
      |                                        VERSION_STR
make: *** [Makefile:250: main.o] Error 1

martinwhitaker, 2025-05-23:
Please provide more details. How did you obtain the code - git clone or downloaded tarball? Are you using the latest code from the master branch? Are you building in or out of tree? What commands did you run prior to `make check`.

jordankrim, 2025-05-23:
git clone https://github.com/steveicarus/iverilog.git
mv iverilog iverlog-12.0
cd iverilog-12.0
sh autoconf 
./configure --prefix=<my home>
make check

martinwhitaker, 2025-05-23:
Running this on Mageia 9 with gcc 12.3.0, after correcting the typo, gives:
````
% git clone https://github.com/steveicarus/iverilog.git
Cloning into 'iverilog'...
...
% mv iverilog iverilog-12.0
% cd iverilog-12.0
% sh autoconf
% ./configure --prefix=/home/me
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for gcc... gcc
...
% make check
mkdir dep
Using git-describe for VERSION_TAG
g++ -DHAVE_CONFIG_H -I. -Ilibmisc  -Wall -Wextra -Wshadow   -g -O2 -std=c++11 -MD -c main.cc -o main.o
mv main.d dep/main.d
g++ -DHAVE_CONFIG_H -I. -Ilibmisc  -Wall -Wextra -Wshadow   -g -O2 -std=c++11 -MD -c async.cc -o async.o
mv async.d dep/async.d
...
````
So on your RHEL 8 system `make` appears to be ignoring the dependencies on `dep` and `version_tag.h`.

What happens if you run `make version`?

jordankrim, 2025-05-23:
Using git-describe for VERSION_TAG
/bin/sh: line 3: /opt/rh/rh-git218/root/usr/bin: No such file or directory

martinwhitaker, 2025-05-23:
Do you get the same error if you run `git describe` in the `iverilog-12` directory?

jordankrim, 2025-05-23:
No.  I get this:
s20250103-36-gea26587b5

martinwhitaker, 2025-05-23:
In your top level `Makefile`, search for the line starting `@if $(GIT_PRESENT)` and remove the `@`, then run `make version` again. That should show what commands it is actually trying to execute.

jordankrim, 2025-05-23:
If still does not seem to work for some reason.  Instead, I downloaded the zip file (thus avoiding git altogether) and it compiles and installs.

caryr, 2025-06-19:
Have you been able to debug this build failure further? I compile from git on the LTS versions of Ubuntu, BSD and Windows all from git with no issues.

caryr, 2025-06-19:
Ok, I will close this for now since it appears to not be an `iverilog` issue.