steveicarus/iverilog #1253: Assignment pattern to unpacked array using variables rejected at elaboration

1353369570, 2025-06-19:
### Test Case

```systemverilog
module test(output logic y [2:1]);

  logic wire1 = 1'b1;
  logic reg1  = 1'bz;

  assign y = '{wire1, reg1};  // assignment pattern with variables

endmodule

Compilation Command：

iverilog -g2012 test.sv -o a.out

Actual Output

test.sv:6: sorry: Continuous assignment of assignment pattern is not yet supported.
1 error(s) during elaboration.

Expected Behavior
This is valid SystemVerilog syntax. The compiler should either:

Accept it and simulate correctly, or Emit a more informative error if assignment patterns to unpacked arrays are not fully supported.

### Summary

Continuous assignment of an unpacked array using an assignment pattern of **variables** is currently **rejected during elaboration** with a "not yet supported" error, whereas using **constants** results in a different failure during simulation.

This reveals an inconsistency in support for assignment patterns in Icarus Verilog.

---

caryr, 2025-06-19:
This is a duplicate of #654 and `iverilog` is clearly reporting this is not currently supported.

caryr, 2025-07-01:
This code elaborates and runs properly using the development branch. It does fail using the stable branch (V12).