steveicarus/iverilog #1254: Simulation error in `always @*` block ?

toby-bro, 2025-06-19:
When simulating this module I would have expected out to be `1` whereas it is simulated as `0`.

```bash
iverilog -o module_sim_iv -g2012 -gsupported-assertions ./testbench.sv && ./module_sim_iv
```

I get this output
```
out has value: 0
./testbench.sv:36: $finish called at 20 (1s)
```

```sv
module always_at_star (
    input bit [7:0] in,
    output bit out
);
    always @* begin
        out = ~in[0];
    end
endmodule

module top;
    logic [7:0] in;
    logic out;

    always_at_star dut (
        .in(in),
        .out(out)
    );

    initial begin
        string line;
        int fd;
        int status;

        in = 8'h00;

        #20;

        $display("out has value: %b", out);

        $finish;
    end
endmodule
```

When I remove the `always @*` and put an assign instead I do not have the issue.
Is this an obscure but legit behaviour of `always @*`  that I am unnaware of or is this a bug ?

toby-bro, 2025-06-19:
Ok. Thanks for the answer. 

Nevertheless I couldn't help but think that this still seemed a surprising behaviour as I would have thought that the default value of `out` would have been `X` at the start of the program and the fact that I assign a value to it would have been thus considered a transition. Especially as iverilog supports `X` and `Z` values.

toby-bro, 2025-06-19:
By the way the other simulators I tried this code on all gave a `1` (yosys, cxxrtl, verilator, slang...). (I know that icarus won't be bullied into changing it's behaviour but thought I'd mention it 😅 )

caryr, 2025-06-19:
The default value of the logic `out` variable at the top is `1'bx`, but it has an implicit continuous assignment from the `in` variable inside the `always_at_start` module which is a `bit` and has a default value of `1'b0` to the module port, so the top-level `out` is assigned `1'b0`. Assigning a new value to the sensitivity list that does not create a transition should not execute the block. An `always_comb` block is the only block that automatically executes at time-zero to resolve these kind of issues.

If you use the limited sysnthesis functionality (`-S`) inside `iverilog` to translate this event based code to combinational logic then you do get the result you are expecting. Also changing the module to use`logic` ports solves the issue since now the x to 0 transition is detected on the input.

FYI a recent version of a commercial tool returns the same result, so I believe `iverilog` is working as expected and the other tools are the ones with the bug. Strength in numbers is only valid if you have not all made the same mistake. If `yosys` or any of the other tools are synthesizing this to gates then you would expect it to return `1'b1`, but as coded `iverilog` is returning the result expected by the standard.

toby-bro, 2025-06-19:
Thank you very much for your clear answer !

caryr, 2025-06-20:
I'm going to close this since `iverilog` is working as expected.

closed 2025-06-20 by caryr