steveicarus/iverilog #1257: Arrays not supported for VCD/FST dumping?

pweaverm, 2025-07-03:
Hello,
  I'm working on some System Verilog modeling and am trying to use the following 

logic [1:0] delay_stage [9];

Unfortunately, in the VCD file, this array signal isn't listed at all. I tried seeing if it was broken apart somehow or renamed but it seems to be not dumped. The same result seems to be in the FST dump as well.

Are multi-dimensional arrays supported?

Thanks

Mark

caryr, 2025-07-04:
VCD has no support for directly dumping unpacked arrays. and since we use the same commands to dump VCD or FST I've never looked to see if this could be added to just the FST dumper. It's also slightly dangerous to automatically dump unpacked arrays since they have the possibility of being very large in size.

The following will dump the individual words of an unpacked array, though it generates a warning that can safely be ignored if you are not using a conflicting escaped identifier:
```
module top;
  logic [1:0] delay_stage [9];

  initial begin
    $dumpvars(0, top);
    for(int idx=$low(delay_stage); idx<=$high(delay_stage); ++idx) $dumpvars(0, delay_stage[idx]);
  end
endmodule
```
This scheme could likely be embedded directly in the dumper to handle explicitly adding the unpacked array variable to the `$dumpvars()` command, but we will likely never automatically dump arrays because of the possibility they could have a large size. We should also look for conflicting names and report an error only if there is a real conflict, but this is complicated since there is an escaped identifier created for dumping the array element.

pweaverm, 2025-07-04:
Hi Cary,
 This is interesting because in Xcelium I do this all the time. Cadence has a dedicated waveform database (shm) so maybe that is the difference. There is also a probe variable to limit the array size to save.
  Will Icarus ever have its own database and viewer? Arrays make the code so much more compact and readable. I'm was not so concerned about size since this was not an issue with Xcelium.

Appreciate the quick response and possible work around.

Thanks

Mark

caryr, 2025-07-04:
Proprietary dumpers and formats can do whatever they want. I should look at FST and see if it supports arrays in some manner, but you have to make sure the sensitivity list is correct since you ideally only want to update the word that changed instead of the entire array anytime an element changes. It's really the waveform viewer that determines the format not the simulator.

We'll likely never implement a waveform viewer.

caryr, 2025-07-04:
Also if FST supports or adds support for array dumping an array would never be dumped implicitly from a scope, but if you added them explicitly to the `$dumpvars()`  that could be supported and then it's the users responsibility to make sue they do not dump somethings too big.

pweaverm, 2025-07-07:
Hi Caryr,
  Adding the $dumpvar() explicitly for arrays is okay. It would be good to be able to write code with mulit-dimensional arrays and be able to plot the results. I'm not sure how this would be more of an issue regarding size than individual registers.

For example:
if I have 4 - 8bit words a1, a2, a3, a4 and wanted to create a shift register delaying an 8bit input

**Without arrays this would be written:**
#in a clock process
a1 <= in;
a2 <= a1;
a3 <= a2;
a4 <= a3;
assign out = a4;

**With arrays:**
logic [7:0] a [5];

#in a clock process
a <= {in, a[0:3]};
assign out = a[4];

I don't know, but do both of these result in the same size of waveform dump? If so, then being able to plot the array a would be great. Have no idea is FST can do this but I am surprised that VCD does not have it.

Agree, that a warning or a user defined limit for array size dumping would be a good idea.

Thanks!

Mark

caryr, 2025-07-07:
If you can find where VCD defines support for arrays please let me know. I skimmed the latest version of the standard and I don't see where array support is defined.

For your example there is no difference, but memories can have many more elements and they may not all be updated at the same time so the sensitivity is different. If you treat arrays as just a single block the entire contents of the array would be dumped any time a single element changes. We need something that recognizes arrays as a group, but allows updating only the elements that change at a given time point. The trick I showed you does this since they are saved as individual elements.

Multidimensional arrays are even more tricky since the waveform viewer would need help knowing how to plot the various dimension.

I know if you dump a VCD file using a specific commercial simulator it will also complain about arrays not being supported. Though it's not very clear it's specifically arrays that are not supported.

pweaverm, 2025-07-07:
Thanks Cary!

I will start to add the workaround in my code. I think my understanding is just from my work, with Xcelium and Simvision. We use arrays for a lot of signals since the data path is N words by M bits. Most of the memories are hard macro IP which also have an array for the model.

Understand and agree, that if the array has a single word changing, it's probably not a good idea to store the whole array in the waveform database each time. I'm not to familiar with the VCD spec, I would have thought that it supported toggles on any RTL structure since VCD can be passed to Voltus for power calculations.

Thanks for the help and promoting to an enhancement. 

Mark

martinwhitaker, 2025-07-07:
My probably faulty recollection is that NCverilog would dump small arrays to a VCD file by dumping each word using an escaped identifier, e.g. `\my_array[0]`, etc.

I agree this would be a useful enhancement.