steveicarus/iverilog #1258: How to suppress 'warning: Timing checks are not supported.' warnings?

martinwhitaker, 2025-07-05:
### Discussed in https://github.com/steveicarus/iverilog/discussions/1207

<div type='discussions-op-text'>

<sup>Originally posted by **zapta** January 26, 2025</sup>
When we run the command below, we get tens fo 'Timing checks are not supported.' warnings. Is there a flag or a similar way to suppress them, without modifying the source file or running post processing filters? 

```
iverilog -g2012 -o _build/blinky_tb.out -DVCD_OUTPUT=_build/blinky_tb -I"/Users/user/.apio/packages/oss-cad-suite/share/yosys/gowin" "/Users/user/.apio/packages/oss-cad-suite/share/yosys/gowin/cells_sim.v" blinky.v blinky_tb.v

/Users/user/.apio/packages/oss-cad-suite/share/yosys/gowin/cells_sim.v:176: warning: Timing checks are not supported.
/Users/user/.apio/packages/oss-cad-suite/share/yosys/gowin/cells_sim.v:190: warning: Timing checks are not supported.
/Users/user/.apio/packages/oss-cad-suite/share/yosys/gowin/cells_sim.v:191: warning: Timing checks are not supported.
/Users/user/.apio/packages/oss-cad-suite/share/yosys/gowin/cells_sim.v:207: warning: Timing checks are not supported.
/Users/user/.apio/packages/oss-cad-suite/share/yosys/gowin/cells_sim.v:208: warning: Timing checks are not supported.
/Users/user/.apio/packages/oss-cad-suite/share/yosys/gowin/cells_sim.v:226: warning: Timing checks are not supported.
/Users/user/.apio/packages/oss-cad-suite/share/yosys/gowin/cells_sim.v:227: warning: Timing checks are not supported.
/Users/user/.apio/packages/oss-cad-suite/share/yosys/gowin/cells_sim.v:228: warning: Timing checks are not supported.
/Users/user/.apio/packages/oss-cad-suite/share/yosys/gowin/cells_sim.v:246: warning: Timing checks are not supported.
/Users/user/.apio/packages/oss-cad-suite/share/yosys/gowin/cells_sim.v:247: warning: Timing checks are not supported.
...
```

The first warning is about the $setup() function below
```
(* abc9_flop, lib_whitebox *)
module DFF (output reg Q, input CLK, D);
        parameter [0:0] INIT = 1'b0;
        initial Q = INIT;

        specify
                (posedge CLK => (Q : D)) = (480, 660);
                $setup(D, posedge CLK, 576);
        endspecify
                
        always @(posedge CLK)
                Q <= D;
endmodule
```
</div>

martinwhitaker, 2025-07-05:
Fixed in the master branch.

closed 2025-07-05 by martinwhitaker

zapta, 2025-07-06:
Thanks @martinwhitaker!