steveicarus/iverilog #1263: Failed to elaborate port default value.

1353369570, 2025-07-24:
when "iverilog -v -g 2012 "  the following design :

 ```
module dut #(
    parameter int DEFAULT_WIDTH = 8
)(
    input logic [DEFAULT_WIDTH-1:0] data_in = '0,
    output logic [DEFAULT_WIDTH-1:0] data_out
);

    assign data_out = data_in;

endmodule
```

The ecpect result: 
Successfully finished

But the  actual result is: 

[image]

caryr, 2025-07-27:
Also, it's unlikely that the code provided matches the code that generated the message. It looks like the message is from code that is using a port default that is a parameter for a top level module. That issue is reported in #1202 and parameter defaults for non-toplevel ports was recently fixed for cases except for instance arrays, see #1230 .

I'm not certain why people all of a sudden think it's appropriate to add ports to top level modules. Yes, it's allowed in the standard, but makes no sense from a usage standpoint.

caryr, 2025-07-27:
Please provide a complete example that demonstrates the issue