steveicarus/iverilog #1266: Trireg is not supported

L3odr0id, 2025-08-02:
Compilation of the following SystemVerilog module throws an error:
```SystemVerilog
module a(output trireg logic b);
endmodule: a
```

Command:
```
iverilog -g2012 test.sv
```

Error output:
```
test.sv:1: syntax error
test.sv:1: Errors in port declarations.
```

Expected behavior:
This module should be compiled.

This issue occurs in `Icarus Verilog version 13.0 (devel) (s20250103-44-gf82c6c7b3)`

Found by a [verilog model](https://github.com/DepTyCheck/verilog-model) using [deptycheck](https://github.com/buzden/deptycheck)

martinwhitaker, 2025-08-02:
This is documented in the README. Do you actually have a need for `trireg`?

[I notice the README is out of date in other respects. There was more up-to-date information on the old Wiki, which really needs to be pulled across]

caryr, 2025-08-02:
This is from some kind of framework that while different than fuzzing is still not from code people are actually trying to model a design with. It's also documented as missing in a SourceForge ticket. I think we need to do a more detailed list of what is not supported and then link that to specific tickets. I've closed a couple duplicate tickets recently. I have not verified this is valid code since it is not warning about trireg not being supported. It could be a corner case we did not cover initially or could be invalid.

L3odr0id, 2025-08-03:
> This is documented in the README.

Oh, sorry, I missed that. I don't want to report features that are documented as unsupported