steveicarus/iverilog #1267: Wire net treated as an uwire

L3odr0id, 2025-08-05:
Compilation of the following SystemVerilog module throws an error:
```SystemVerilog
module a(input uwire logic a1);  
endmodule: a

module b();
  wire logic b1;

  a b_inst(.a1(b1));
  not not_inst(b1, b1);

  assign b1 = 'b0;
endmodule: b
```

Command:
```
iverilog -g2012 test.sv
```

Error output:
```
test.sv:5: vvp.tgt error: (implicit) uwire "b1" must have a single driver, found (2).
error: Code generation had 1 error(s).
```

Expected behavior:
This module should be compiled. `b1` is declared as `wire logic`, but it looks like it's being treated as `uwire` instead.

This issue occurs in `Icarus Verilog version 13.0 (devel) (s20250103-44-gf82c6c7b3)`

Found by a [verilog model](https://github.com/DepTyCheck/verilog-model) using [deptycheck](https://github.com/buzden/deptycheck)

caryr, 2025-08-05:
It actually looks like the compiler thinks this is a variable that is then implicitly cast to a uwire and that resulting net is driven by both the output of the inverter and the continuous assignment which is a violation of the single driver rule for variables/uwire. It appears that the compiler is interpreting `wire logic b1` as a variable declaration.

caryr, 2025-08-06:
The standard clearly states that `wire logic` and `wire` should be identical net declarations.