steveicarus/iverilog #1269: Release more often?

ktbarrett, 2025-08-15:
The v12_0 was the last release and it is 2.5+ years old at this point. It includes a number of bugfixes. Bugfixes that remain unresolved in OSes that package iverilog because it isn't in a release yet. Many projects, even rather slow ones, tend to do at least one release a year, or even two. Are releases very difficult or is there something preventing doing a minor version release once or twice a year?

martinwhitaker, 2025-08-17:
See also discussion #1058. I have always been in favour of more frequent releases, but I can never persuade the other Icarus developers.

Minor releases are problematic because they would require bug fixes to be back-ported to the stable branch. I used to do that as a matter of course for all bugs I fixed that didn't require an API change, but the changes to the test suite broke my flow, so I gave up on that.

caryr, 2025-08-17:
I think we would all like to release more often! From my current perspective I have a couple fixes in flight, but have not had time to work on finishing them. I tend to favor a release around the end of the year since I often have some time off and can finish up what I'm working on and get ready for a release, but it's a fairly narrow window.

I think part of the slowness is also related to our long term stability in the development branch and then if you want the latest bug fixes/features you just need to compile your own from the development branch and if people are doing that it seems to take off some of the pressure to actually perform a release? When I was actively using Icarus for real work I always used the development branch because of the fixes and newer functionality and that has likely lowered my prioritization of releases.

I have a few things I would like to see in the next release so maybe this all needs to go into #1058 to give better focus and clarity regarding what will and will not be in the next release.

caryr, 2025-08-17:
FYI my general goal for a release is to fix as many of the bugs as we can, but this last year it has been harder to keep up since we have had both fuzzing and frame work people submitting corner case bugs that while valid may not be encountered by a normal user writing code or they would just work around the current limitation.

ktbarrett, 2025-08-19:
Ah yeah. That would be nice. We do have a way around it that's basically discovering everything greedily. As long as your design is small, it's not an issue. Also *ALL* VHPI simulators have the same issue (besides NVC). I wouldn't worry about it too much.