steveicarus/iverilog #1271: Assignment to real throws syntax error

L3odr0id, 2025-09-02:
Compilation of the following SystemVerilog module throws an error:
```SystemVerilog
module a(output real wreal);
  assign wreal = 1.0;
endmodule: a
```

Command:
```
iverilog -g2012 test.sv
```

Error output:
```
test.sv:2: syntax error
test.sv:2: error: Syntax error in left side of continuous assignment.
```

Expected behavior:
it seems real support is not that important, but the message is misleading. It should state that real is not supported in continuous assignments.

This issue occurs in `Icarus Verilog version 13.0 (devel) (s20250103-64-gdad78d525)`

Found by a [verilog model](https://github.com/DepTyCheck/verilog-model) using [deptycheck](https://github.com/buzden/deptycheck)

L3odr0id, 2025-09-02:
Oh. I didn't understand the issue at first. 

The module
```Systemverilog
module a(output real w1);
  assign w1 = 1.0;
endmodule: a
```
compiles correctly. The problem occures only if vairable's name is `wreal`. For some reason exactly this name causes the problem

L3odr0id, 2025-09-02:
And this module
```systemverilog
module a(output logic wreal);
  assign wreal = 1.0;
endmodule: a
```
has no real type declared, but throws same error as the first example
```
test.sv:2: syntax error
test.sv:2: error: Syntax error in left side of continuous assignment.
```

martinwhitaker, 2025-09-02:
`wreal` is a Verilog-AMS keyword. It is also recognised when the Icarus Verilog extended types are enabled, although I can't find that documented anywhere. To work round this, compile with the `-gno-xtypes` option.

L3odr0id, 2025-09-02:
Wow. Thanks for the explanation!

caryr, 2025-09-10:
We need to think if there is anything we can do to help make it obvious this is a correct error message in the context of the extended types, but I'm not sure how to categorize this so leaving it blank for now.