steveicarus/iverilog #1275: Assertion failed: type_ == FUNC, file ../iverilog/net_scope.cc, line 535

shirriff, 2025-09-09:
When I run iverilog on this very broken code, I get "assertion failed":
```

shirriff, 2025-09-09:
When I run iverilog on this (very broken) code, I get `Assertion failed: type_ == FUNC, file ../iverilog/net_scope.cc, line 535`:

```
module latch();
endmodule
module mc133(output a);
  assign a = latch();
endmodule
```

Expected behavior: I assume that it should give me an error message instead of an assertion failure.

I'm just submitting this bug in case you want to know about problems with assertions. Feel free to close it if it's unimportant.

Environment:
Icarus Verilog version 12.0 (devel) (s20150603-1539-g2693dd32b)
Windows 11

caryr, 2025-09-09:
When I checked this with the 12 release and the latest development both report the following:

br_gh1275.v:4: error: No function named `latch' found in this context (mc133).
br_gh1275.v:4: error: Unable to elaborate r-value: latch()
2 error(s) during elaboration.

Can you please double check this code matches what is causing the assertion.

caryr, 2025-09-09:
Maybe you can ask them to package the actual 12.0 release or a newer development release in the 13.0 line.