steveicarus/iverilog #1280: task $fmonitor() is not currently implemented

FSharp4, 2025-09-28:
When trying to run verilog testbenches that contain the `fmonitor` command I get the following output:

```
SORRY: <file>:<line> task $fmonitor() is not currently implemented
```

Example:
```verilog
/*
 * test_fmonitor.v
 * Tests fmonitor command support (for Icarus Verilog)
 */
`timescale 1ns/10ps
module test_fmonitor;
    reg A;
    integer f;
    initial begin
        f = $fopen("test_fmonitor.out");
        $fmonitor(f, "At t=%d, A = %b", $time, A);
        A = 1;
        #10;
        A = 0;
        #10;
        $fclose(f);
        $finish;
    end
endmodule;
```

In shell:
```
$ iverilog -o test_fmonitor test_fmonitor.v
$ ./test_fmonitor           
SORRY: test_fmonitor.v:11: task $fmonitor() is not currently implemented.
```

Replicated on macOS and Windows. There was a prior issue about this that was closed (#592), and I wasn't able to find another issue on this, so I'm filing this one.

FSharp4, 2025-09-29:
Documented as not implemented yet in vpi/sys_icarus.c

martinwhitaker, 2025-09-29:
I don't know why `$fmonitor` wasn't implemented. Possibly because the method used to disable `$fstrobe` when `$fclose` is called is not very robust and was considered to be more dangerous with `$fmonitor`.

caryr, 2025-09-30:
If I remember correctly it was that `$monitor()` only supports a single active instance while `$fmonitor()` supports multiple active instances and supporting multiple instances will required a rewrite of the base functionality.

closed 2025-10-05 by martinwhitaker